tralamazza/micropython #60: examples: Fail to enter raw repl

xmeow, 2017-06-03:
When I try to execute some example like 
`python pyboard.py test.py --device COM5`
the script return fail to enter raw repl. I got a pca10040 and success compile and enter normal repl mode.
How you guys run the examples? Cause they seem so delicious~

glennrub, 2017-06-03:
This is how I do it.. 
Either I freeze my modules/scripts by moving them into the freeze/ folder, or I use the REPL "paste mode" (CTRL-E). 

We should probably take a look on how to get the raw repl up again if it is useful. I remember removing it from main.c main loop long back.

glennrub, 2017-06-03:
Hi @xmeow, 

I have added back the raw REPL mode in PR #62. 

Honestly, i do not know how this raw REPL is suppose to work. Would you be able to help me verify this? It looked like you had an idea on how to use it :)

xmeow, 2017-06-04:
Thank you so much, the freeze way works fine for me. But the pyboard.py script or ampy still return could not enter raw repl. I'm not sure if it is caused by disabling the flow control. I will dig further into this. Anyway thank you so much for such grate work.

closed 2017-06-04 by xmeow