tralamazza/micropython #84: UART access example

BWhitten, 2017-06-14:
Could you give a basic UART usage in the examples.
I have been unable to read from the UART in the python script using the following:
```
serial = UART(id = 0, baudrate = 115200)
while True:
        data = serial.readline()
        if data != None:
            print(data)
```

glennrub, 2017-06-14:
I did a test with UART REPL (to free up the instance), and I did also see some issue with readline(). I will investigate this a bit further.

For now, read()/writechar() seemed to work quite well with Putty. 

    >>> from machine import UART
    >>> serial = UART(0, baudrate=115200)
    >>> serial.writechar(ord('h'))
    >>> chr(serial.readchar())
    'a'
    >>> chr(serial.readchar())
    'b'

Tracing the readline implementation, it goes back to mp_stream_p_t.read which points to machine_hard_uart_read. This is the same c-function called by the serial.readchar() in the sample I posted above. I will see if i can get a debugger up tonight and see where it leads.

glennrub, 2017-06-14:
I believe i have the answer now. 
To make uart.readline() work you need to terminate the line using \n (10) and **not** \r (13). Most terminals like _putty_ and _screen_ sends \r.

You can see the termination criterea here: 
https://github.com/tralamazza/micropython/blob/master/py/stream.c#L414

Here is the PC code i tested successfully with:

    import serial
    serial = serial.Serial(port="/dev/ttyUSB0", baudrate=115200, rtscts=True)
    serial.write("This is a long test...\n")
    serial.close()

closed 2018-07-19 by glennrub