tralamazza/micropython #95: nrf/boards/microbit: Backport persistent filesystem module

glennrub, 2017-08-13:
Depends on hal/hal_nvmc to be implemented.

Could driver be put under nrf/boards/microbit/extra_drivers or similar and have a new boards drivers hook to be compiled in?

Reference to micro:bit source: https://github.com/bbcmicrobit/micropython/blob/master/source/microbit/persistent.c

aykevl, 2017-10-09:
How should this be implemented?
To me it seems most logical to integrate it into the VFS system. That makes it much more flexible.: it can be put on any block device. Maybe it's best to put it in MicroPython itself - some other chips might also want to use this type of filesystem.

glennrub, 2017-10-09:
This task was originally intended to replicate the same microbit API, being a direct user of hal_nvmc when available. The idea was as well to expose the backported module to other targets (nrf51 and nrf52). But, you have a point when it comes to VFS.

The flash footprint of using raw hal_nvmc access will be significantly less then adding in FAT/VFS. So this is one of the targets where it makes sense to strip out FAT/VFS, as the original port did not have this. However, to also make microbit as a device a bit more attractive and updated on API (machine API, BLE support, FAT/VFS etc.) i have been thinking that this target (microbit) should probably have one more compile flag, LEGACY=1, to distinguish between legacy mode and modern API. So i would say this backport task belongs to the legacy mode. However, the tiny flash access module could potentially be reused by other targets.

As you understand, i have not thought this through fully. I'm very open to suggestions here.

aykevl, 2017-10-09:
(I updated my comment above before I saw your update)

I did a test and it turns out FAT requires somewhere between 20 and 50kB of storage. With a SoftDevice, there is only 10kB. So using FAT for on-chip storage is impossible on the nRF51+SoftDevice.

I see a few options (more combinations possible):
* keep VFS, blockdev and FAT but add the microbit filesystem directly talking to hal_nvmc. This may be useful for systems that might add SD card support but also want to use what's left of internal storage.
* drop both VFS and blockdev, and re-implement `open`, `os.listdir`, `os.remove`, etc.

I think it may be best to port the microbit filesystem over to hal_nvmc, and if it makes sense maybe integrate it with VFS after that. Each board may then be configured to either use an SD card only (with FAT), use internal storage only (with microbit filesystem), or both. I'm hesitant about adding SD card support by default (and thus FAT) when a board doesn't have an SD card adapter.

glennrub, 2017-12-02:
Merged in PR #126. Can be enabled by setting MICROPY_HW_HAS_BUILTIN_FLASH (1) and HAL_NVMC_MODULE_ENABLED.

closed 2017-12-02 by glennrub