tralamazza/micropython #96: nrf/hal: Add hal_nvmc persistent file access HAL

glennrub, 2017-08-13:
In order to save data directly to the flash on the nrf51/nrf52 the nvmc should be implemented.

Requirements:
- Should handle 1k (nrf51) and 4k (nrf52) page size.
- Should allow to operate while Bluetooth LE stack or other radio stack is active (has to use sd_nvmc instead of cmsis nvmc function call in the case of active radio). 
- Should add a clean linker script entry to set the region used as data storage.

aykevl, 2017-10-07:
Here's an initial attempt at support: https://github.com/tralamazza/micropython/compare/master...aykevl:nrf-nvmc-wip
It's very much a work in progress, but I've been able to use `open`, `uos.listdir` etc. and even do imports from locally stored .py files. It's mostly inspired on the ESP8266.
To set up, I needed to format the data first using `uos.VfsFat.mkfs(mount.bdev)` - apparently FAT support was already included (on all boards, including boards that don't have an SD card slot).

If this is the right approach, I can finish it (including SoftDevice support - if there is any room left).

glennrub, 2017-10-08:
Awesome! Thanks for picking this one up! :-)

I took a look at the code you have written so far, and it looks very promising. 
I will not be able to try it out before a few days, but i'm really looking forward to it :-)

Yes, your observation is correct, FAT on all boards, even if not all have SD card slot. The reasoning for this was that the nrf5x devices have quite limited flash, and given the possibility of adding the sdcard.py as a frozen module to provide an extended storage space using SPI would improve this. When that said, i believe it takes about 15-20k ROM. So it could be worth a patch to facilitate compile time enable/disable of it in the future. 

The big challenge is probably around the SoftDevice support. I'm not sure how familiar on using the softdevices, so i will give some pointers in any case, which might be educational for others as well.

When the softdevice is enabled, direct flash access using NRF_NVMC will halt the CPU while writing/erasing. The CPU might be halted longer than a BLE radio interval and disconnect the link. Therefore, all access to flash has to go through the softdevice so that the access is queued through the softdevice, and let the softdevice find a sweet time to perform the action. The result will come a bit later, through an interrupt event. If i recall this correctly, its sharing the IRQ handler with the BLE stack (ble_drv.c). If the softdevice is present and not enabled, the sd_flash* functions will return a retval immediately after the operation has been performed, and no event will be given.
I guess it all ends up in that we have to move the IRQ handler from Bluetooth and make it common, dispatching BLE events to ble_drv.c and ble_soc events to flash handler. 

Would it make sense to split this task into two. First add native support for flash like you have started on, then create a new task adding support for SD flash access? What do you think?

aykevl, 2017-10-09:
15-20k is **huge**. For the nRF51 devices with a SoftDevice replacing it with the microbit filesystem would save a lot: there is almost no room left currently. FAT32 support could later be added again using native module support (WIP, https://github.com/micropython/micropython/issues/3311). But for SoftDevice-less devices and the nRF52 it's less of an issue, it may even be too much space for the tiny microbit filesystem (it currently supports about 30kB, while the nRF51 without SoftDevice has ~110kB space left).

Your description of the NVMC on a SoftDevice is about the same as what I've read about it. It's a bit more complex so that's why I left it out initially. I would need to take a look to see how difficult it is.
https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s110.api.v8.0.0%2Fgroup___n_r_f___s_o_c___f_u_n_c_t_i_o_n_s.html

Yes I agree it may be best to split this task. In a first version I could keep things roughly as they are now, moving FlashBdev to native code to (hopefully) reduce code size and adding some code to format/mount the filesystem automatically on startup. After that, SoftDevice support can be added and (if you agree) the FAT32 filesystem replaced with the smaller microbit filesystem on nRF51 devices with SoftDevice.

I hope to have time for this in the following days.

aykevl, 2017-10-09:
I've moved FlashBdev into the nrf module and added automatically formatting and mounting of the filesystem. I'm not too happy about the literal Python string in main.py, but it costs a little less (24 bytes) than moving it into a separate frozen _boot.py file like on the ESP8266.
I've also added support for the WT51822-S4AT as that's the only device I can test on - but that's only to show how it can be enabled: the build fails with a SoftDevice enabled.
If it looks good I can make a pull request.

glennrub, 2017-10-09:
I'll try to disable all the filesystem files and see what number i end up with. I think this is an important number forward.

glennrub, 2017-10-09:
I did some ugly disable of VFS/FAT here, trying to keeping the compiler in, as this might be needed in any case of a files system, raw or not.  Reference to the work: https://github.com/glennrub/micropython/commit/a51345826c232bdae1e126740a575526e6b43ec8

This is the test results i got: 

With VFS/FAT:
```
   text	   data	    bss	    dec	    hex	filename
 165524	   1424	  36176	 203124	  31974	build-pca10040/firmware.elf
 142760	    248	  13520	 156528	  26370	build-pca10001/firmware.elf
```

Without VFS/FAT: 
```
   text	   data	    bss	    dec	    hex	filename
 148588	   1424	  35656	 185668	  2d544	build-pca10040/firmware.elf
 125500	    248	  13000	 138748	  21dfc	build-pca10001/firmware.elf
```

Approx usage of ROM: 
NRF51: 17260
NRF52: 16936

glennrub, 2017-10-10:
Hi @aykevl. 

I got an idea when driving back home from work today. I was thinking about this user defined linkerscript region in your branch proposal above. Wouldn't it be better if this went automatically? To simply use what is left. 
Imagine that there is a symbol marking the end of use flash (text, data, ..),  "end_used_flash".

I'm thinking something like this: 
_flash_filesystem_start = ALIGN_TO_UPPER_PAGE(end_used_flash);
_flash_filesystem_length  = LENGTH(FLASH_TEXT) - end_used_flash;

I'm not really sure if it is possible to align to 1024, 4096 in a linker script.. However, if this is not possible, it can be aligned in software.

What do you think? Is there any other reason for keeping flash potentially unused that i haven't thought of?

aykevl, 2017-10-11:
Thank you for the VFS/FAT stats!

About using unused flash space: yes, I've considered it. It shouldn't be that hard, and it's certainly possible to align in a linker script (it has most expressions that also exist in C).
The reason to keep it fixed is so that the start page doesn't move automatically with updates, which would otherwise corrupt the filesystem requiring a format - removing all files that currently exist. Of course, that's a trade-off between using all available space and not (unintentionally) removing stored files.

---

I've ported the micro:bit filesystem over, integrating it in the VFS. That means both filesystems can be used. A usecase for this is bootstrap code to mount an SD card (not sure how that's usually done?). It's still a bit rough around the edges, but appears to work fine. Unfortunately, it isn't possible to import files from there (yet), as that's [not yet implemented in the VFS](https://github.com/tralamazza/micropython/blob/master/extmod/vfs.c#L122).
I've also implemented initial support for writing with the SoftDevice enabled, but it isn't finished yet (doesn't wait for the callback, which works as long as the BLE stack is disabled).

Currently the microbit filesystem uses ~2kB of TEXT, leaving 8kB for actual file storage. I suspect FAT is actually the bigger part of VFS+FAT, so disabling it should increase the storage quite a bit. Otherwise it might also be possible to remove VFS altogether (as a compile time switch) like you did, as there isn't much use to a VFS without FAT and just one mounted filesystem.

glennrub, 2017-10-12:
I been thinking a bit about this need to format the FATFS. I don't know FAT that very well, but would it be possible to create a pre-formated FS partition on PC side, and place it in the FLASH_USER area using two flash runs (or merge formated FS into the hex)? This way, it would be possible to preserve the data or "reformat" it during flashing, and no need for runtime python script on this part. As I said, i'm not an expert on FAT, so it might be a terrible idea.

aykevl, 2017-10-12:
Yes, that should be possible. I'm not sure how difficult it would be, though - maybe it's possible to generate a FAT image using the cross compiler or a host utility like `mkfs.vfat`. But maybe it isn't as straightforward. The resulting binary can be included in a .hex using either `objcopy` (no idea whether that's possible) just like `objcopy` is currently used to extract firmware.hex from firmware.elf, but the other way round. Another way would be to include the image in a C source file as a big array (like frozen Python modules) and be placed in a special section that's then moved into the flash user area with some linker script instructions.
(offtopic: it may also be possible to include the SoftDevice in the final .hex file in a similar way, requiring one less step to flash a device).

But why do it? I agree that it's quite ugly how it is done right now, with the inline Python script in `main()`. It feels like a hack. But it should be possible to do the same using C code, probably even reducing code size (see d2d6702734eabe3214771c70e7dcc6f4589c06e7). That removes the Python script.
An advantage is providing initial data in the FAT image (boot.py and/or main.py) - but then again, I don't know whether that's worth the effort.

---

I think I'm almost done with the filesystem support. I fixed imports and filesystem modifications while the SoftDevice is running. However, it's still not finished and some parts are less than ideal (especially around where to place code/modules and [naming things](https://martinfowler.com/bliki/TwoHardThings.html)). TODO: the microbit filesystem without VFS, testing [ampy](https://github.com/adafruit/ampy) and running `boot.py`/`main.py` on startup. If you have some initial feedback that would be nice.

glennrub, 2017-10-12:
Yeah, you are probably right, it could make it even more complicated having external fatfs format and merge/double flash. I guess this should also probably be a solution that comes from upstream, not invented in this port... 

------------

I'll see if i can set of some time to start testing your branch a bit.

Just one heads up on the ampy; i guess you will use this with uart without flow control (wt51822-s4at). There is a known issue when it comes to UART when not using flow control #89. This issue might also hit when using ampy. If the device is busy doing flash operations (like saving to flash/anything blocking RX) RX packets might be dropped without the sender's notice. In the REPL case its loosing RX bytes due to TX is blocking. I'll say we can address ampy as a follow up.

aykevl, 2017-10-15:
I think I'm basically done. Supported:

* microbit filesystem - with `os.listdir()` and `os.remove()` (no `os.size()` yet because I wondered whether we should include such a nonstandard API). With and without VFS support. Question: where to place the code for this? The uos module seems a bit out of place, although it is related.
* Using FAT on the internal flash
* Importing scripts from microbit filesystem with VFS disabled
* Running `main.py` on startup if it exists, runs file from sdcard if mounted - question: should we also run `boot.py`?

I can make a pull request if that's easier. Diff: https://github.com/tralamazza/micropython/compare/master...aykevl:nrf-nvmc-wip

The only thing that's missing is ampy support, it requires a line ending with "soft reboot" to be printed after a soft reboot. Otherwise ampy works fine (with cleared flash, a very low baudrate and a very small file, at least).

```diff
--- a/ports/nrf/main.c
+++ b/ports/nrf/main.c
@@ -241,6 +241,7 @@ pin_init0();
     if (ret_code == PYEXEC_FORCED_EXIT) {
         NVIC_SystemReset();
     } else {
+        printf("PYB: soft reboot\n");
         goto soft_reset;
     }
```

glennrub, 2017-10-15:
Awesome @aykevl, this is some great work! I did actually try out the microbit filesystem a bit yesterday, from your work branch. It worked very well :-) 

When it comes to your questions, i don't really have all the answers...
1. Where to place the code: As the mbfs_obj is never exposed in any module_globals_table, directly mapped in main.c i see your point. However, it could also be it's own module in modules/ folder, for example, "modules/microbitfs/"?
2. Running boot.py/main.py: Looking at the stm32 port, it looks like it is expected to create a boot.py if it did not exists, mount the SD card, then try to run boot.py and main.py. I guess it would be good to align with the other ports as flash memory is now available! :-) 

Yeah, the ampy support. I do not mind adding such a print. I see this could be a subject of change itself, as it has the prefix "PYB". However, a quick search in the repo shows that all ports uses this prefix. If you would like to add it, could you send it as a seperate PR?

The way forward.. I do have some initial feedback, there is a few things i noticed: 
- The changes done in extmod/vfs are needed, but i find it hard to merge them into this port. I would really like to keep the diff against upstream only on the ports/nrf/ folder, and let all changes outside, come from through upstream repo. Should we try to get these changes on extmod/vfs through upstream?
- A small change in microbitfs.c is needed to support softdevices. The randomise_start_index() needs to use hal_rng as NRF_RNG is a reserved peripheral when SD is activated.

I'll use next week to help test it a bit, providing some more feedback :-) I could also help out to test and update the nrf52 target board files if you would like me to.

aykevl, 2017-10-15:
> 1. Where to place the code: As the mbfs_obj is never exposed in any module_globals_table, directly mapped in main.c i see your point. However, it could also be it's own module in modules/ folder, for example, "modules/microbitfs/"?

In the end, I didn't know whether that was a much better place so I kept it were it is now. Either way doesn't really matter to me.

> 2. Running boot.py/main.py [...]

I've added boot.py for consistency.
I'm not so sure about putting initial files there on format. They take a bit of space (and there isn't a lot of space anyway), and we don't expose those files to the outside world like the pyboard does (on the SD card). So if we provide initial contents for those files, IMHO there must be a good reason other than just imitating what other ports do.

About ampy: I'll send a pull request after this one is done. Note that the zephyr port just prints "soft reboot\n" without the "PYB: " prefix.

> * The changes done in extmod/vfs are needed, but i find it hard to merge them into this port. [...]

I've sent a pull request upstream: https://github.com/micropython/micropython/pull/3371
I can remove it from here, as it is only needed when both VFS and the microbit filesystem are enabled.

> * [...] randomise_start_index() [...]

Fixed.

> I could also help out to test and update the nrf52 target board files if you would like me to.

If you want to, that would be great! I also can't test many other boards - they might not work out of the box if they have more modules enabled (requiring an adjusted linker script).

glennrub, 2017-10-16:
I have started to update all the other boards configuration, and started to test. I'll put up a test plan here to give some indication on where i am in the testing and status. 

Current diff against your WIP branch is here: https://github.com/aykevl/micropython/compare/nrf-nvmc-wip...glennrub:fs_testing

nrf51 targets:
[ ] pca10000 - microbitfs
[ ] pca10000 - microbitfs + SD
[ ] pca10000 - hal_nvmc fat + VFS
[ ] pca10000 - hal_nvmc fat + VFS + SD
[ ] pca10001 - microbitfs
[ ] pca10001 - microbitfs + SD
[ ] pca10001 - hal_nvmc fat + VFS
[ ] pca10001 - hal_nvmc fat + VFS + SD
[ ] pca10028 - microbitfs
[ ] pca10028 - microbitfs + SD
[ ] pca10028 - hal_nvmc fat + VFS
[ ] pca10028 - hal_nvmc fat + VFS + SD
[ ] pca10031 - microbitfs
[ ] pca10031 - microbitfs + SD
[ ] pca10031 - hal_nvmc fat + VFS
[ ] pca10031 - hal_nvmc fat + VFS + SD
[ ] microbit - microbitfs
[ ] microbit - microbitfs + SD
[ ] microbit - hal_nvmc fat + VFS
[ ] microbit - hal_nvmc fat + VFS + SD

nrf52 targets
[ ] arduino_primo - microbitfs
[ ] arduino_primo - microbitfs + SD
[ ] arduino_primo - hal_nvmc fat + VFS
[ ] arduino_primo - hal_nvmc fat + VFS + SD
[ ] feather52 - microbitfs - **IN PROGESS** - failing due to wrong location of TEXT_USER or garbage data in mounted flash.
[ ] feather52 - microbitfs + SD
[ ] feather52 - hal_nvmc fat + VFS
[ ] feather52 - hal_nvmc fat + VFS + SD
[ ] pca10040 - microbitfs
[ ] pca10040 - microbitfs + SD
[ ] pca10040 - hal_nvmc fat + VFS
[ ] pca10040 - hal_nvmc fat + VFS + SD
[ ] pca10056 - microbitfs
[ ] pca10056 - microbitfs (no softdevice available yet)
[ ] pca10056 - hal_nvmc fat + VFS (no softdevice available yet)

Not able to test:
[ ] wt51822 - microbitfs
[ ] wt51822 - microbitfs + SD
[ ] wt51822 - microbit - hal_nvmc fat + VFS
[ ] wt51822 - microbit - hal_nvmc fat + VFS + SD
[ ] dvk_bl652 - microbitfs
[ ] dvk_bl652 - microbitfs + SD
[ ] dvk_bl652 - hal_nvmc fat + VFS
[ ] dvk_bl652 - hal_nvmc fat + VFS + SD

aykevl, 2017-10-17:
A few notes on the feather52 you're testing:

- A lot has changed in the flash system in the nRF52 (according to release notes), so it is possible some more changes are needed. I didn't notice any changes in a quick read of the reference manual, though(now called the Objective Product Specification).
- You can try erasing the whole device before flashing the new firmware - maybe there is something left in the remaining space that gives trouble (IIRC when the first byte of a block is 0xff it assumes the page is cleared, which it may not be).
- are you sure putting the user flash at address 0 is correct? It wouldn't be on the nRF51.
  Also see for example (check the version!): https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk52.v0.9.1%2Fbledfu_memory.html

Oh and if you didn't know you can create checkboxes using ` - [ ]` and ` - [x]`.

 - [ ] unchecked
 - [x] checked

glennrub, 2017-10-17:
Thanks @aykevl, for noticing the bug i made. A bit to late for writing linker scripts i guess. It works now, after setting the right start address of FLASH_USER.

The idea with feather52 is to preserve the Serial-DFU and SD the device its shipped with, and use the  Serial-DFU as programming interface for users.

aykevl, 2017-10-17:
> Which i interpret as: (f_frsize * f_blocks) 512 * 30 = 15360.

A quick calculation on my device confirms this (writing a 1024-byte file consumes two blocks of 512 bytes - this pattern continues with 10 of such files).
But if it really is just 15kB free space, I think that would be really unfortunate. That means there is about 49kB or ~75% overhead. The tiny microbitfs might be a better fit then - it's just 128 bytes per file (for the name) + 2 bytes per block of 128 bytes. It does introduce some overhead (~2kB together with VFS+FAT) but even that is much less than the FAT overhead. FAT would then be reserved for SD cards, not the internal FS.
My own device (with 512*69=35328 usable bytes) uses 103kB in storage, so is also very inefficient (67% wasted).
Note that I already included some spare pages - I just calculated them, there are 9 pages left (SD-less, 134kB text while 152kB is reserved).

**Summary**: it turns out that FAT32 is really inefficient and maybe it's a better idea to just use the microbit filesystem even without SoftDevice. Besides you have a good point about leaving some space - it may come in use in the future. Maybe we could tune the microbit FS to be able to use bigger areas (60kB should be easy) or it might be possible to shrink the FAT32 space somehow. But for now the microbit filesystem seems to be a better choice.

---

Assuming this is without a SoftDevice.
That's really weird - I've never experienced it. I've only seen it when I forgot to close the file. It's possible that the area wasn't formatted correctly, possibly because you were trying different sizes of FAT filesystems. Can you try this?

```python
os.umount('/')
os.VfsFat.mkfs(nrf.flashbdev)
os.mount(nrf.flashbdev, '/') # or reset
```

I tried to mount with a different filesystem size, and the filesystem mounts successfully without formatting first. I've also tested the exact commands you did (writing foobar.py), and it worked well in my case - after formatting the filesystem.

glennrub, 2017-10-17:
I tried to erase the chip to make sure everything is wiped, then do the unmount, mkfs, and now it gets stuck in the mkfs (pca10028). I'll try to get into a debug mode on this one tomorrow, and see where its halting.

I did a gdb run of it to see where its halted, now during mkfs (address 0x000186c8).
firmware.map snippet:
```
 .text.hal_nvmc_write_words
                0x00000000000186a0       0x4c build-pca10028/hal/hal_nvmc.o
                0x00000000000186a0                hal_nvmc_write_words
 .text.machine_reset
                0x00000000000186ec       0x1c build-pca10028/modules/machine/modmachine.o
```
From what i can read its halted in the middle of hal_nvmc_write_words.

aykevl, 2017-10-17:
Can you try with the following patch applied?

```patch
--- a/ports/nrf/hal/hal_nvmc.c
+++ b/ports/nrf/hal/hal_nvmc.c
@@ -112,10 +112,19 @@ bool hal_nvmc_erase_page(uint32_t pageaddr) {
     return true;
 }
 
+#include "py/runtime.h"
+
 bool hal_nvmc_write_words(uint32_t *dest, const uint32_t *buf, size_t len) {
     // Note that we're writing 32-bit integers, not bytes. Thus the 'real'
     // length of the buffer is len*4.
 
+    if ((uintptr_t)dest & 3) {
+        mp_raise_ValueError("dest is not aligned");
+    }
+    if ((uintptr_t)buf & 3) {
+        mp_raise_ValueError("buf is not aligned");
+    }
+
     // Configure NVMC so that writes are allowed (anywhere).
     NRF_NVMC->CONFIG = NVMC_CONFIG_WEN_Wen;
     while (NRF_NVMC->READY == NVMC_READY_READY_Busy) {}
```

glennrub, 2017-10-18:
I tried out the patch this morning, and no different result. It looks like there is no issue with the alignment. 

------------

However, i managed to get a debug target up running on nrf52 to get it running, and see if there was something that could hint on the nrf51 issue.

It turned out that the nrf52 device does cannot create the FATFS with 64k allocated FLASH_USER. Debugging with show that i hit https://github.com/tralamazza/micropython/blob/master/lib/oofatfs/ff.c#L5144. 

Experimenting a bit with the FLASH_USER sizes, tracing the sz_vol and i got this matrix:
FLASH_USER 64k, sz_vol = 12
FLASH_USER 192, sz_vol = 48
FLASH_USER 224, sz_vol = unknown, as reflash triggers a reboot which successfully formats it. However, the VFS + FAT + NRF is working as expected. Open, write, close does now work.

The surprising part was when i did a os.statvfs("/"). Its even less blocks allocated then on nrf51 with 64k configured. I have used the working setting of 224k FLASH_USER: 
```
>>> os.statvfs("/")
(512, 512, 22, 22, 22, 0, 0, 0, 0, 255)
```
Summarized i have it working on nrf52, but i cannot explain the numbers.

glennrub, 2017-10-18:
Got the nrf51 target working (pca10028), bug in my linker script setting incorrect start address of FLASH_USER. However, my comments above is valid. That we need to at least reserve 50 pages to make format work correctly, both on nrf51 and nrf52. Is it possible to expand the block size of FAT to align with page sizes, to utilize the whole page?

aykevl, 2017-10-19:
I think you got it right - that's the same conclusion I reached. FAT requires at least 50 pages, or it won't even format.

It's this property (looked it up in the esp8266 config):

    #define MICROPY_FATFS_MAX_SS           (1024)

With it, I was able to use the full page size of the nRF51. But it still requires 50 pages, which is quite some space on the nRF52 (200kB). And at least on the nRF51 with a page size of 1024 it adds 18 pages of overhead (the file allocation table itself, I guess), apparently independent of the page size.

Note: defining the page size does add some extra size, about 500 bytes (because oofatfs now cannot rely on one fixed page size).

---

I've played around some more with the FAT filesystem. It looks like it uses at least one block per file, even files with one char in it. Additionally, it is possible to reduce the FAT overhead a bit, namely by changing `n_rootdir` to e.g. 128 (overhead reduced to 6 pages with pagesize 1024, I think 2 are fixed and 4 are for the rootdir). `n_rootdir` cannot be changed to less than this value for a blocksize of 4096 (or we'd end up with no rootdir at all!) - see https://github.com/tralamazza/micropython/blob/master/lib/oofatfs/ff.c#L5088.

So it's nice FAT works on nRF5x internal storage, but I'm getting less and less exited about actually using it - the microbitfs seems like a better (default) choice.

---

Somewhat unrelated: support for importing from non-FAT filesystems should be merged after MicroPython 1.9.3 is released (see https://github.com/micropython/micropython/pull/3371#issuecomment-337126821).

aykevl, 2017-10-28:
I see you did quite some tests already.

> - [ ] pca10028 - microbitfs + SD - **FAILED** during open("testing.py", "w"). BLE link still in connection, but console blocked, probably due to while loop in main thread.

Ah, yes. I suspect this is a deadlock between the NVMC waiting for the write/erase operation to finish and the SoftDevice waiting for the BLE REPL (which is blocked by the NVMC operation) to receive a char. I should be able to debug this (but not right now) - but I'm not entirely sure how to fix it.

> - [ ] pca10028 - microbitfs + VFS + External SD card + SD **FAILED**

I guess because there was no space left for .text?

---

I've changed the microbit filesystem configuration a bit so it can use up to 64kB of space. This may be useful on nRF52 devices with it's bigger page sizes. Currently in a separate branch as I didn't know where to put these commits:

https://github.com/aykevl/micropython/compare/nrf-nvmc-wip...aykevl:nrf-mbfs-bigger

aykevl, 2017-10-31:
>￼ pca10028 - microbitfs + SD - `FAILED` during open("testing.py", "w"). BLE link still in connection, but console blocked, probably due to while loop in main thread.

I cannot reproduce this. I have successfully created a file `hello.py` with contents `print("hi!")` using the WebBluetooth console (but had to work around the ~20-char limit).

glennrub, 2017-10-31:
I'm sorry the testing takes so much time...   

Last weekend I ended up in Linker script tuning for the feather52, which set me a bit back. It looks like that the target generates data outside the defined .text region. I did brute force attempts to see how big images that could be transferred over DFU, which ended up in a max FLASH_TEXT size of 0x267c0 (157.69KB). Theoretically it should have been 162KB (app + ISR). I have tested this in an earlier phase and it did pass. However, that was without BLE REPL. Basically, i cannot transfer the DFU image with BLE REPL + microbitfs, even if i can make it link and come below 162KB. So, it could be some error in the memory layout on the feather52 or i have some kind of garbage data in the hex file outside the .text region which is interpreted as application size when generating DFU images from hex. I'm not done investigating here.

---------

>> ￼ pca10028 - microbitfs + SD - FAILED during open("testing.py", "w"). BLE link still in connection, but console blocked, probably due to while loop in main thread.

> I cannot reproduce this. I have successfully created a file hello.py with contents print("hi!") using the WebBluetooth console (but had to work around the ~20-char limit).

 I have not started to look into this yet, only added a note to myself that this should be checked up further. I just went ahead to test the next targets. I used the exact scripts i list in the testing post above and it fails on the open(). However, i'm not ruling out that this might be a bluetooth driver bug, and not really related to the flash access. That you also did the same test, which passes, makes that suspicion stronger. Thanks for trying to replicate this one! 

>> Note: Overlapping VFS partition with others. Could nrf be mounted as "/flash" in VFS case?

> I only see this now. While testing I couldn't enable MICROPY_HW_HAS_SDCARD (sdcard_is_present does not exist, among other errors) - am I missing something?

I will try to clarify this one a bit. Quickly summarized i'm thinking that "/" could be free, and mounts could have a name. Like "/sd", "/flash" etc. instead of having "/sd" as a folder in "/". When writing this note, i was hitting a mount issue where "/" overwrote my "/sd" mount by mounting file system at the root after SD Card. I cannot recall how i triggered this one...

glennrub, 2017-12-02:
Merged in PR #121. Can be enabled by setting HAL_NVMC_MODULE_ENABLED in boards' nrf51_hal_conf.h/nrf52_hal_conf.h.

closed 2017-12-02 by glennrub