tralamazza/micropython #108: nrf/drivers/bluetooth: s110 not working using web bluetooth console in Ubuntu/Debian

glennrub, 2017-10-08:
frozen test.py on pca10001: 
```
from ubluepy import Service, Characteristic, UUID, Peripheral, constants
from pyb import LED

def event_handler(id, handle, data):
    print("BLE event:", id, "handle:", handle)
    print(data)

    if id == constants.EVT_GAP_CONNECTED:
        # connected
        LED(2).on()
    elif id == constants.EVT_GAP_DISCONNECTED:
        # disconnect
        LED(2).off()
    elif id == 80:
        print("id 80, data:", data)

# u0 = UUID("0x180D") # HRM service
# u1 = UUID("0x2A37") # HRM measurement

u0 = UUID("6e400001-b5a3-f393-e0a9-e50e24dcca9e")
u1 = UUID("6e400002-b5a3-f393-e0a9-e50e24dcca9e")
u2 = UUID("6e400003-b5a3-f393-e0a9-e50e24dcca9e")
s = Service(u0)
c0 = Characteristic(u1, props = Characteristic.PROP_WRITE | Characteristic.PROP_WRITE_WO_RESP)
c1 = Characteristic(u2, props = Characteristic.PROP_NOTIFY, attrs = Characteristic.ATTR_CCCD)
s.addCharacteristic(c0)
s.addCharacteristic(c1)
p = Peripheral()
p.addService(s)
p.setConnectionHandler(event_handler)
p.advertise(device_name="micr", services=[s])
```

Driver log from run in Android: 
```
>>> import test
Is enabled status: 0
SoftDevice enable status: 0
IRQ enable status: 0
BLE enable status: 0
Is enabled status: 0
Is enabled status: 0
Is enabled status: 0
Is enabled status: 0
Device name applied
encoded uuid for service 0: 9e ca dc 24 0e e5 a9 e0 93 f3 a3 b5 01 00 40 6e
ADV: uuid size: 16, type: 20, uuid: 1, vs_idx: 2
Set Adv data size: 27
>>> GAP CONNECT
BLE event: 16 handle: 0
None
GAP CONN PARAM UPDATE
GAP CONN PARAM UPDATE
GAP CONN PARAM UPDATE
GATTS write
BLE event: 80 handle: 14
bytearray(b'\x01\x00')
id 80, data: bytearray(b'\x01\x00')
GATTS write
BLE event: 80 handle: 11
bytearray(b'a\r')
id 80, data: bytearray(b'a\r')
GATTS write
BLE event: 80 handle: 11
bytearray(b'b\r')
id 80, data: bytearray(b'b\r')
```

Driver log from Ubuntu:
```
>>> import test
Is enabled status: 0
SoftDevice enable status: 0
IRQ enable status: 0
BLE enable status: 0
Is enabled status: 0
Is enabled status: 0
Is enabled status: 0
Is enabled status: 0
Device name applied
encoded uuid for service 0: 9e ca dc 24 0e e5 a9 e0 93 f3 a3 b5 01 00 40 6e
ADV: uuid size: 16, type: 20, uuid: 1, vs_idx: 2
Set Adv data size: 27
>>> GAP CONNECT
BLE event: 16 handle: 0
None
GAP CONN PARAM UPDATE

*STUCK*
```

aykevl, 2017-10-15:
(continuing from #107)
I can confirm the issue with WebBluetooth is related to the MTU size - I've managed to manipulate the `Pin` output (in short messages) but sending large messages crashes the connection.

glennrub, 2017-10-27:
Have not yet tried it out, but this might be interesting, maybe it works better:
https://forum.micropython.org/viewtopic.php?p=22782#p22782

closed 2017-11-05 by glennrub