tralamazza/micropython #115: FLASH_TEXT overflow

jhill1440, 2017-10-21:
I am trying to flash micropython to a feather nrf52 and am using the vagrant machine/setup as described in the youtube video by tony D.  I am going through it step by step and keep getting this error when issuing "make BOARD=feather52"

CC modules/music/musictunes.c
CC modules/ble/modble.c
CC modules/random/modrandom.c
CC build-feather52/pins_gen.c
LINK build-feather52/firmware.elf
/usr/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: build-feather52/firmware.elf section `.text' will not fit in region `FLASH_TEXT'
/usr/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: region `FLASH_TEXT' overflowed by 704 bytes
collect2: error: ld returned 1 exit status
make: *** [build-feather52/firmware.elf] Error 1

If I try to do "make BOARD=feather SD=s132" it does the same thing but the overflow is greater.  Instead of 704 bytes it is 13040 bytes.

If I do not specify a board and just issue "make" it finishes fine.  But the it gets hung up when doing "make SD=s132 dfu-gen" . Probably because the default board is not DFU compatible.

Is there and easy was to increase the FLASH_TEXT memory size? 

Thanks
Joshua

glennrub, 2017-10-21:
Hi Joshua, 

You are absolutely right! The .text region is over-filled. I suspect this is a side-effect of enabling floating point support on the nrf52 targets (#86). Some bytes have to be optimized away to make it work again as expected. For now, the quick-fix would be to disable floating point support on you target in order to free up about 23 Kbytes. 

You can do this in the mpconfigport.h, https://github.com/tralamazza/micropython/blob/master/ports/nrf/mpconfigport.h#L55, setting the define to MICROPY_FLOAT_IMPL_NONE as the nrf51 targets.

Cheers,
Glenn

jhill1440, 2017-10-21:
Hello,

Thanks for the quick answer.  Unfortunately that did not work.

If I changed mpconfigport.h line 53:
-MICROPY_FLOAT_IMPL
+MICROPY_FLOAT_IMPL_NONE

I still get the overflow error.

If I changed mpconfigport.h line 55:
-MICROPY_FLOAT_IMPL
+MICROPY_FLOAT_IMPL_FLOAT

I get all sorts of errors.

If I change both, I get the same errors as just changing line 55.

I tried going into:
ports/nrf/boards/feather52/custom_nrf52832_dfu_app.ld
and add memory to line 10:

`-FLASH_TEXT (rx) : ORIGIN = 0x0001c400, LENGTH = 0x026c00 /* 152 KiB - APP - ISR */`
`+FLASH_TEXT (rx) : ORIGIN = 0x0001c400, LENGTH = 0x04d800 /* 152 KiB - APP - ISR */`

This would double the memory allocated to FLASH_TEXT.

I can get it all to compile but the when flashing to the feather it fails.  It looks like it is going to start then fails out.  I think the memory is not allocated correctly since I just went and arbitrarily messed with it.

I am a real noob when it comes to this stuff but I don't mind leaning and failing.

Hope this helps with what is wrong.

Thanks
josh

jhill1440, 2017-10-21:
Hello,

If I remove the lines added in:
https://github.com/tralamazza/micropython/pull/86/files

the 5 lines added specifically to mpconfigport.h.  Once removed it compiled fine.

and then compiles fine with `make BOARD=feather52 SD=s132`

`CC modules/ble/modble.c
CC modules/random/modrandom.c
CC build-feather52-s132/pins_gen.c
LINK build-feather52-s132/firmware.elf
   text	   data	    bss	    dec	    hex	filename
 148820	    344	  19964	 169128	  294a8	build-feather52-s132/firmware.elf
arm-none-eabi-objcopy -O binary build-feather52-s132/firmware.elf build-feather52-s132/firmware.bin
arm-none-eabi-objcopy -O ihex build-feather52-s132/firmware.elf build-feather52-s132/firmware.hex`

and then do the:
`make BOARD=feather52 SD=s132 dfu-gen`
`make BOARD=feather52 SD=s132 dfu-flash`

and it works!!!!!!!!!!!!!

glennrub, 2017-10-21:
I'm glad to hear that you made it work! :-)
I'll leave this issue open until it is fixed, trying to free up some more flash space by default.

glennrub, 2017-10-21:
Feather52 memory layout: https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/memory-map

Proposal for changes in custom feather52 linker script:
* ISR length = 1k = 0x400
* TEXT start = SD + App ISR = 0x1c400
* TEXT length = 162kb - 1k ISR = 161kb = 0x28400

This should give the application (161kb (new) - 155kb (today)) = 6k more to Application .text. After this the is cap reached. 
For file storage #96, the spare area of 28kb set of for application storage can be used. start = 0x6d000, length = 0x7000.

glennrub, 2017-10-22:
Hi Joshua,

I have prepared a patch for the feather52 you could try out, #117.

I have tested it locally here, and it works for me. I have updated the custom linker script with what i believe is the max size it can allow. After increasing the .text region it was still lacking 7k flash, so i added a default disablement of floating point for this target. If you would like to get in floating point, you will have to manually disable other features. However, i believe this could work as an OK default target configuration atm.

It would be great with some feedback if this works out for you as well.

Cheers,
Glenn

jhill1440, 2017-10-24:
Hello,

I tried using the patch with DISABLE_FLOAT = 'yes'

and it works great.  no problems at all.

Thanks
Joshua

aykevl, 2017-11-01:
I tried an even lower ISR vector table on the nRF51 (0x100 or 256 bytes), and it seemed to work fine. I did this by reducing the size FLASH_ISR in the linker script and extending FLASH_TEXT to it. On the nRF51 it uses just 168 bytes (42 interrupts * 4 bytes) so 856 bytes can be saved.
This may be helpful for the nRF52 as well.

glennrub, 2017-11-15:
Optimizations have made its way into the master branch, and the feather52 target should not have any .text overflow anymore. Closing this issue.

closed 2017-11-15 by glennrub