tralamazza/micropython #116: main.py like file

jhill1440, 2017-10-21:
Is there a file that can be loaded on a micropython flashed feather nrf52 that will be executed at start up?  Could be loaded afterward or bundled with it during make. Right now we just have a prompt (>>>) that can execute code but nothing is saved.  Is there a way to save code on the microcontroller?

THanks
Joshua

glennrub, 2017-10-21:
There is some work in progress which relates to your question about file storage #96. 

Another way to get the python files over, is to put them in the freeze/ folder. Then the modules will be compiled (frozen) into the hex file. If you want to load the module at startup you could for example modify the lines in main.c to make it load your module that you have frozen.
https://github.com/tralamazza/micropython/blob/master/ports/nrf/main.c#L177

There is already a test.py file you can play with in the freeze folder. You could try load the module in REPL or in main.c and see how it works.

aykevl, 2017-11-21:
As far as I know this should be solved.
You can put a main.py file in freeze/ and if I remember correctly, I've tested this and it works.

Additionally, you can enable MICROPY_HW_HAS_BUILTIN_FLASH and set HAL_NVMC_MODULE_ENABLED, see this commit (ignore MICROPY_PY_NRF):
https://github.com/aykevl/micropython/commit/2520c4790a28d5158cf9583ff76a557a3c4997a3
This will enable the internal filesystem, but requires #133 to actually send files to it.

closed 2018-07-19 by glennrub