tralamazza/micropython #119: compile pca10040, got error: stack usage computation not supported for this target

linycm, 2017-11-01:
Hello,

      When I compile pca10040 in Unbutu (under VM), I got the result :"
../../py/nlrthumb.c: In function 'nlr_push':
../../py/nlrthumb.c:80:1: error: stack usage computation not supported for this target [-Werror]
"
Could you help? Thanks

Rdgs,
Martin

glennrub, 2017-11-03:
Hi Martin, 

I managed to reproduce the exact same issue as you have here, by using a windows PC with an old compiler, gcc-4.9.3 version. 
However, it works for me in Linux using gcc-5.4.1 and gcc-6.3.1.

I'm not sure what compiler you have used, maybe you could tell a bit more on what environment you have? gcc version, ubuntu version?

Or, just go directly on testing a fresh compiler: 
https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads

Unpack it and set "export PATH = /my/new/gcc/path:$PATH
Then run the compilation again...

Cheers,
Glenn

linycm, 2017-11-05:
Hi Glenn,

  Thank you for your help! I got it work when I change to arm-none-eabi-gcc 6.3.1. Before I think I am using gcc version 5.4.0 20160609 (Ubuntu16.04.4) , in fact the system use arm-none-eabi-gcc  4.9.3 since I installed before.

  I don't know how to let system compile using gcc 6.3.1, the system default compiler is 4.9.3. So I add cross compiler address as below and work:
sudo make CROSS_COMPILE=/(new gcc path)/gcc-arm-none-eabi-6-2017-q2-update/bin/arm-none-eabi-  SD=s132

Rdgs,
Martin

glennrub, 2017-11-15:
Hi Martin,
 
Is this still an issue? Or has it been resolved by updating the compiler?

Cheers,
Glenn

linycm, 2017-11-18:
It‘s resovled. thanks!

glennrub, 2017-11-21:
Great! :)

closed 2017-11-21 by glennrub