tralamazza/micropython #159: Custom board modules not built, Makefile issue (only on OSX?)

tdicola, 2018-02-23:
I noticed the Makefile unfortunately isn't handling custom board modules like with the microbit board.  It's silently skipping the board module build and hiding a lot of problems (none of the board modules are being built).  This is with building on OSX so it might be a quirk of just this POSIX system (I'm not much of a bash and make expert to know for sure).  But here's the issue and how to verify yourself.

1. In the nrf folder make with BOARD=microbit.

2. Everything seems to build fine (however if you actually load the hex on a microbit you'll see it doesn't have the microbit module inside it).

3. Comment out this check in the main Makefile: https://github.com/tralamazza/micropython/blob/master/ports/nrf/Makefile#L29 so that the board modules make include is forced in:
```
#ifeq ($(shell test -e boards/$(BOARD)/modules/boardmodules.mk && echo -n yes),yes)
	include boards/$(BOARD)/modules/boardmodules.mk
#endif
```

4. Re-build the microbit port, make BOARD=microbit.

5. Unfortunately now you'll see the build fail because the board modules are now being compiled:
```
tony-macbook:nrf tony$ make BOARD=microbit
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Generating build-microbit/genhdr/mpversion.h
mkdir -p build-microbit/boards/microbit/modules
CC boards/microbit/modules/microbitdisplay.c
boards/microbit/modules/microbitdisplay.c: In function 'microbit_display_show_func':
boards/microbit/modules/microbitdisplay.c:73:11: error: 'MP_QSTR_image' undeclared (first use in this function); did you mean 'MP_QSTR_wake'?
         { MP_QSTR_image,    MP_ARG_REQUIRED | MP_ARG_OBJ, {.u_obj = MP_OBJ_NULL} },
           ^~~~~~~~~~~~~
           MP_QSTR_wake
boards/microbit/modules/microbitdisplay.c:73:11: note: each undeclared identifier is reported only once for each function it appears in
...
```
(many more QSTR complaints)

So it looks like two issues:

1. The Makefile logic for including board specific modules is problematic, at least on OSX.  I'm not much of a make or bash expert but this check appears to be silently failing and masking more issues:
```
ifeq ($(shell test -e boards/$(BOARD)/modules/boardmodules.mk && echo -n yes),yes)
	include boards/$(BOARD)/modules/boardmodules.mk
endif
```
If it helps just trying the test command on my OSX machine returns yes (with no newline) as expected:
```
tony-macbook:nrf tony$ test -e boards/microbit/modules/boardmodules.mk && echo -n yes
yestony-macbook:nrf tony$
```
However for whatever reason the make check is failing and not going into the board include, as shown by forcing the board include on with comments.

2. It appears the microbit build might be broken with the qstr issues.  It's probably an easy fix to regen qstrs and separate from this Makefile issue.  I'll play around more to see what's up and if it's just not running the qstr generation.

Anyways FYI it looks like some weirdness with the Makefile and board specific modules.  The version of make I'm using too if it helps is:
```
tony-macbook:nrf tony$ make --version
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-apple-darwin11.3.0
```

tdicola, 2018-02-23:
For what it's worth too I tried with the latest 4.2 version of make from homebrew and get the same results, the ifeq check is failing to include board modules.  Make info:
```
tony-macbook:nrf tony$ gmake --version
GNU Make 4.2.1
Built for x86_64-apple-darwin15.6.0
Copyright (C) 1988-2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

tdicola, 2018-02-23:
So the -n in echo appears to be the problem, if I add this info before the check to see what's happening:
```
$(info Check: $(shell test -e boards/$(BOARD)/modules/boardmodules.mk && echo -n yes))
```
I see this output:
```
Check: -n yes
```
So for whatever weird reason bash is echoing the -n instead of interpreting it as an option.  This explains why the ifeq is failing since it's comparing '-n yes' to 'yes'.  If I change the ifeq check to look like (remove the -n):
```
ifeq ($(shell test -e boards/$(BOARD)/modules/boardmodules.mk && echo yes),yes)
	include boards/$(BOARD)/modules/boardmodules.mk
endif
```
It now works as expected and goes on to compile the board modules (and fails, but that's a separate thing and probably easy fix).

Very weird--like I said I'm really not a bash expert and have no idea why it's behaving strangely.  No idea if this change would impact building on Linux or other systems.

tdicola, 2018-02-23:
Yeah qstr errors were a red hering, I just needed to clean and then remake with the ifeq fix applied.  With regenerated qstrs it went on and built all the board modules for the first time just fine.  However I now get errors linking but will dig and file a separate issue if it looks necessary.

aykevl, 2018-02-23:
Note that `test` is both a built-in of bash and a systemwide binary. The Makefile probably doesn't pick the bash built-in but instead uses the systemwide `test` binary, which apparently gives an issue on OS X.

aykevl, 2018-02-23:
Possible solution:
https://stackoverflow.com/questions/8346118/check-if-a-makefile-exists-before-including-it

glennrub, 2018-02-23:
You are right @aykevl, the trick from the link you sent works. I'll prepare a patch.

glennrub, 2018-02-26:
Hi @tdicola,
Does the microbit build target work for you on OSX after PR #161 was merge into master?

glennrub, 2018-04-04:
I'm closing this issue as the referred PR should have solved the issue. If you still have problems please re-open the issue again.

closed 2018-04-04 by glennrub