tralamazza/micropython #160: microbit board build issues

tdicola, 2018-02-23:
I think https://github.com/tralamazza/micropython/issues/159 was masking a few problems with the microbit board.  After I fixed that one I found I also had to make these changes to get the microbit board to build:

(and apologies I can't really send this as a pull right now--my fork is in a weird state as I'm adding support for a new board)

1. This ms variable isn't initialized and a warning is thrown which turns into a build stopping error: https://github.com/tralamazza/micropython/blob/master/ports/nrf/boards/microbit/modules/modmicrobit.c#L46  Easy fix, set ms = 0; and everything is happy.

2. The microbit port needs MICROPY_VFS defined as 1 in its mpconfigboard.h file.  Without it defined the vfs_fat_disk.o and other objects aren't being linked in and you get linking errors.

3. LTO has to be disabled.  I commented out this part of the makefile: https://github.com/tralamazza/micropython/blob/master/ports/nrf/Makefile#L81  The problem is this GCC bug: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=81440 Certain struct usage triggers a warning that can't be disabled correctly in the linker.  I was getting errors in linking that a few structs (mp_tuple_t for clock and arrow graphic collections) were different sizes.  Disable LTO and all is good.  Perhaps the very latest versions of GCC fix this now (I'm using the 7.0 Q4 2017 release but I think only the most recent nightlies have a fix).

aykevl, 2018-02-23:
1. I see, that's a bug (I wonder why we haven't caught that one before). The real fix, when float is disabled, is probably to avoid the check entirely as `mp_obj_get_int` does it internally as well.
2. This is a symptom of having no LTO and no `--gc-sections`. The fix is enabling one of them. See 3.
3. To disable LTO you can try this patch: https://github.com/tralamazza/micropython/pull/158
It will give most of the space savings of LTO while avoiding LTO. LTO can be tricky to get right.

tdicola, 2018-02-23:
Oh interesting, thanks for the tips!  Cool yeah with LTO it seems to boot and get to a REPL that can import stuff.  Haven't done much beyond that though. :)  Will play around and see if it's worth keeping on or disabling with the patch.

Interesting about the FS too.  Yeah I'm up to about 158k of firmware with the microbit modules enabled again and the VFS turned on, which actually isn't too bad.  Definitely worth investigating if full VFS & FAT support is worth the cost.

Kind of related too, one big thing this port doesn't do that the microbit currently does is support a string script file directly embedded at a position in flash memory.  This is how the web editor for the microbit works (https://python.microbit.org/v/1)--the JS on the page takes in a firmware hex file and then concats to it a hex string that embeds the user's program at the last 8k of flash memory (this way the web page doesn't have to try compiling an entirely new firmware from JS :).  A special 'MP' sentinel value is put at the start so in the board main.c startup it can look for that sentinel and run the embedded script instead of looking for a main.py, etc. in the FS (can see their logic here: https://github.com/bbcmicrobit/micropython/blob/master/source/microbit/mprun.c#L117).  So long story short I'm thinking of adding that logic back in so the web editor works with this port.  Since other boards likely won't want or care about this logic I'm thinking of putting it in a board specific define in mpconfigboard.h and then some ifdefs in the main.c can test for it and apply the embedded script logic.  In that case it might make sense to disable and ignore a VFS entirely (it actually still creates a VFS in the tiny space between the end of the firmware and start of that last 8k block, but in practice it's just a few kilobytes at best and can't store much).  Just an FYI I'll be playing with adding it in a fork and could push back up later.  Hoping to do backports of the major missing microbit modules too like the accelerometer, etc. :)

aykevl, 2018-02-23:
Note that VFS/FAT support is usually not necessary, only if you want to connect an external SD card and don't require the ported micro:bit filesystem (they don't work together at the moment, see #135). VFS/FAT is very big, while the micro:bit filesystem is much smaller (about 2kB) and actually works with the internal flash.

In theory there is also a different way, but a bit more difficult from the web editor POV (but should be doable). Instead of putting a file in the last N bytes, we could actually write it to the filesystem itself avoiding the hardcoded 8kB limit (both limiting the size of the script and limiting the filesystem size if the script is small).

glennrub, 2018-02-23:
>  Since other boards likely won't want or care about this logic I'm thinking of putting it in a board specific define in mpconfigboard.h and then some ifdefs in the main.c can test for it and apply the embedded script logic. 

I might have an alternative suggestion to this. Let's assume we keep the main.c generic for all boards, and keep it clean from board specific defines, as boards might come and go. My suggestion is that we can put in a weak function in the main.c for any board to do custom setup if needed. Many boards have inbuilt hardware like accelerometer etc, which might need some kind of initialization. If boards get a weak function to implement this could solve these cases. The weak function could for example be the last step before we start up the REPL loop in main.c. 

example:

main.c

```
__weak void board_specific_init(void) {
    // Do nothing.
}
```

modmicrobit.c

```
void board_specific_inti(void) {
    initailize_accelerometer();
    initialize_compass();
    display_hello_message();
}
```

closed 2018-07-19 by glennrub