tralamazza/micropython #163: Low power in nrf

amaza, 2018-03-04:
I'm trying using nrf as low power as possible.

Using machine.sleep() consumes about 0,47mA.

Putting a __WFE() in ports/nrf/drivers/bluetooth/ble_uart.c makes NUS BLE (after connecting first time) consuming about 1mA.

``int mp_hal_stdin_rx_chr(void) {
    while (isBufferEmpty(mp_rx_ring_buffer)) {
        __WFE();
    }
``
Can be done in uart REPL?

Using nrf51/2  in .c programming, 3-5uA is achievable. It's possible obtain similar values using machine.sleep()? What prevents machine.sleep() going to 3-5uAmps?

Thanks.

amaza, 2018-03-04:
Using sd_app_evt_wait() improves a lot the power consumption (NUS Ble). Also changing MIN/MAX conn interval, and adv interval. With adv interval about 1000ms, and conn interval about 250ms, 2-3uA can be achieved.

In uart REPL, I can't improve to these numbers.

Best regards.

glennrub, 2018-03-05:
Hi @amaza, thanks a lot for the feedback on the BLE REPL and your findings about power consumption. The points you are making are very good. 

When it comes to connection interval in BLE REPL it is a bit tricky to tell what is the correct values to put in as default here. In the current implementation speed of the BLE REPL has been prioritized over power consuption. However, i can totally see your use-case where you would like to get down the overall power consumption. I'm just thinking that these parameters could fit very well as part of the Peripheral constructor as optional parameters. Then, modify the BLE REPL to be either compiled as power-optimized or speed-optimized as a starter, using pre-compiler defines. At least this is the first that comes to my mind. 

For the advertisment, the default is also quite agressive (62.5ms). Here it could be possible to allow the advertisment interval as part of the ubluepy_advertise_data_t, and update ble_drv.c/ble_drv_advertise_data to accept this interval if set. This way, it should even be possible to start with high frequency advertisement, then stop it, and subsequently re-advertise at a slower rate. It depends a bit on the use-case, but making such API change should make the it a bit more generic and configurable to adjust the overall power consumption.

When it comes to triggering __WFE() or sd_app_evt_wait() from the read path of the BLE REPL, i believe this could be a valuable improvement. I just wonder how this works if a print() is triggered in IRQ path (waking it up from sleep) and then issue a print(). Will the characters come out successfully in the TX path, or will this prevent it from getting back into sleep in RX path?

I totally agree with you points here, and if you have some time for this, i would really encourage you to create some Pull Request step by step on this, such that we can get down this power consumption :)

Cheers

aykevl, 2018-03-28:
The reason why a lot of current is used in the uart REPL is because it currently busy-waits until the next character is received. We could improve on this by using IRQ-based UART, but this hasn't been implemented yet. Now that the new HAL is merged this will be easier to do.

Adding an `sd_app_evt_wait()` should improve things a lot and is easy to fix. I don't think there will be problems with it - AFAIK it wakes up after every interrupt so will instantly see any received chars.

I agree with Glenn that the default advertisement interval is quite short. I'm not sure it is best to push it up to 1000ms by default as it will delay discovery. A quick fix could be to add a 'low power NUS' configuration option that is disabled by default, which limits the advertisement interval and reduces the connection interval. It might even be possible to make this dynamic but that's a bit more work and adds in firmware size.

closed 2018-07-19 by glennrub