tralamazza/micropython #175: build problem on WSL

askvictor, 2018-04-05:
I'm trying to compile for microbit using Windows Subsystem for Linux (ubuntu 14.04) and get the following error - it does the same thing whether I set BOARD=microbit or use the default target. I haven't got a straight linux dev machine handy right now, so it might be WSL issue, or possibly something else in the toolchain - what are the requirements/recommendations for the compiler?

```
$ make BOARD=microbit V=1
python ../../py/makeversionhdr.py build-microbit/genhdr/mpversion.h
CC ../../py/nlrthumb.c
arm-none-eabi-gcc -DBLUETOOTH_LFCLK_RC -DBOARD_SPECIFIC_MODULES -mthumb -mabi=aapcs -fsingle-precision-constant -Wdouble-promotion --short-enums -mtune=cortex-m0 -mcpu=cortex-m0 -mfloat-abi=soft -fno-builtin  -I./boards/microbit/modules -I. -I../.. -Ibuild-microbit -I./../../lib/cmsis/inc -I./modules/machine -I./modules/ubluepy -I./modules/music -I./modules/random -I./modules/ble -I../../lib/mp-readline -I./drivers/bluetooth -I./drivers -I../../lib/nrfx/ -I../../lib/nrfx/drivers -I../../lib/nrfx/drivers/include -I../../lib/nrfx/mdk -I../../lib/nrfx/hal -Wall -Werror -g -ansi -std=gnu99 -nostdlib  -DNRF51 -DNRF51822 -DCONFIG_GPIO_AS_PINRESET -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -fno-strict-aliasing -fstack-usage -Iboards/microbit -DNRF5_HAL_H='<nrf51_hal.h>' -Wl,--gc-sections -ffunction-sections -fdata-sections -Os -DNDEBUG -Os -c -MD -o build-microbit/py/nlrthumb.o ../../py/nlrthumb.c
../../py/nlrthumb.c: In function 'nlr_push':
../../py/nlrthumb.c:83:1: error: stack usage computation not supported for this target [-Werror]
 }
 ^
cc1: all warnings being treated as errors
make: *** [build-microbit/py/nlrthumb.o] Error 1
```

aykevl, 2018-04-05:
I'm trying to reproduce this issue right now, with WSL running Ubuntu Xenial.
I'm suspecting an old version of GCC. You could try removing this line in the Makefile:

    CFLAGS += -fstack-usage

aykevl, 2018-04-05:
I cannot reproduce this on Ubuntu Xenial (inside WSL). I am suspecting you still run Ubuntu 14.04, not the newer 16.04 (Xenial), perhaps from an older WSL installation.
See this post for more details: https://blogs.msdn.microsoft.com/commandline/2017/10/11/whats-new-in-wsl-in-windows-10-fall-creators-update/

askvictor, 2018-04-06:
Just upgraded my WSL from 14.04 to 16.04 and the problem persists; even after a make clean for both mpy-cross and the target. Interestingly, when compiling for the wt51822_s4at board, the error occurs at the link stage, while for microbit it's during a cc.

Removing the -fstack-usage flag does, however, fix this. 

My version of cc is: 
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
```

aykevl, 2018-04-06:
Hmmm I didn't actually let the build finish linking, so maybe that's why it worked for me. I should try it, and try to build the micro:bit target.

I'm happy to hear that removing `-fstack-usage` works for you. I haven't used it myself, maybe it can be removed altogether. @glennrub what do you think?

askvictor, 2018-04-06:
I just spun up a fresh ubuntu 16.04 VM to try building inside that, and get the same problem, so it's looking like a more general error. The compiler is the same version as in WSL (not surprisingly)

aykevl, 2018-04-06:
Yeah WSL runs unmodified Ubuntu binaries so I would be surprised if it behaves differently.

I think it's because Debian packages [gcc-arm-none-eabi version 5.4](https://packages.debian.org/stretch/gcc-arm-none-eabi) while Ubuntu Xenial packages [gcc-arm-none-eabi version 4.9](https://packages.ubuntu.com/search?suite=xenial&keywords=gcc-arm-none-eabi).

askvictor, 2018-04-07:
> Yeah WSL runs unmodified Ubuntu binaries so I would be surprised if it behaves differently. 
I know this is supposed to be the case, but still don't completely trust it yet. It was my first suspect, but is innocent :)

Got debian running WSL, and it compiles fine. Might be worth putting a note in the readme about the required GCC version

aykevl, 2018-05-09:
I think this issue is fixed by #192. It removes `-fstack-usage`.

closed 2018-07-19 by glennrub