tralamazza/micropython #178: cleanup boards mpconfigboard.h

glennrub, 2018-04-10:
Currently many of the defined settings are not used, or overlapping with already existing configs. 
Typical config from pca10040:

```
#define MICROPY_PY_MACHINE_UART     (1)
#define MICROPY_PY_MACHINE_HW_PWM   (1)
#define MICROPY_PY_MACHINE_HW_SPI   (1)
#define MICROPY_PY_MACHINE_TIMER    (1)
#define MICROPY_PY_MACHINE_RTCOUNTER (1)
#define MICROPY_PY_MACHINE_I2C      (1)
#define MICROPY_PY_MACHINE_ADC      (1)
#define MICROPY_PY_MACHINE_TEMP     (1)
#define MICROPY_PY_HW_RNG           (1)

#define MICROPY_HW_HAS_LED          (1)
#define MICROPY_HW_HAS_SWITCH       (0)
#define MICROPY_HW_HAS_FLASH        (0)
#define MICROPY_HW_HAS_SDCARD       (0)
#define MICROPY_HW_HAS_MMA7660      (0)
#define MICROPY_HW_HAS_LIS3DSH      (0)
#define MICROPY_HW_HAS_LCD          (0)
#define MICROPY_HW_ENABLE_RNG       (0)
#define MICROPY_HW_ENABLE_RTC       (0)
#define MICROPY_HW_ENABLE_TIMER     (0)
#define MICROPY_HW_ENABLE_SERVO     (0)
#define MICROPY_HW_ENABLE_DAC       (0)
#define MICROPY_HW_ENABLE_CAN       (0)
```

Could some of these be removed if unused?

glennrub, 2018-04-10:
My proposal would be to keep the port specific ones, deleting all the _HW_ ones inherited from the initial porting.
Also, i would suggest to rename the MICROPY_PY_HW_RNG one to MICROPY_PY_RANDOM_HW_RNG, to indicate which module it is configuring.

As a different task i could image to move LED from pyb module to machine and get a proper MICROPY_PY_MACHINE_LED define.

For non module configs like internal flash, i guess it makes most sense to have something like a _HW_ config (MICROPY_HW_HAS_BUILTIN_FLASH), unless we indicate this a different way, like "MICROPY_PY_OS_INTERNAL_FS"?

Any thoughts?

aykevl, 2018-04-10:
Sounds good!
I once tried to enable `MICROPY_HW_ENABLE_RNG` but it doesn't seem to work. Most other defines of that sort aren't referenced anywhere at all, so can certainly be removed.

glennrub, 2018-05-22:
Proposing PR #202 to move `pyb` to a `board` module. The LED's should be easily deactivated if needed by just configuring the board to have no leds; MICROPY_HW_HAS_LED (0).

glennrub, 2018-06-06:
PR #202 merged. Closing issue.

closed 2018-06-06 by glennrub