tralamazza/micropython #189: nrf52832 s132 5.0.0

dvsseed, 2018-04-20:
I am a newbie at nrf52-dk developer, I changed the file of boards/pca10040/mpconfigboard_s132.mk, from SOFTDEV_VERSION = 3.0.0 to 5.0.0. But after the command ($ make BOARD=pca10040 SD=s132 sd) it has failed to WebBluetooth REPL. So, can someone tell me how to modify the file of boards/nrf52832_512k_64k_s132_5.0.0.ld, or something else. I'd appreciate your help.

glennrub, 2018-04-20:
Hi @dvsseed ,

I did a quick test switching the SD version to v5 for pca10040 and found a compilation and linker error.
I have provided a patch #190 which is working for me to get the BLE REPL up running again.

Reading from the comment, that you have modified mpconfigboard_s132.mk also gives me a hint that you are not on master. Could you try the same using a patched master?

The options to modify using master, provided that the patch #190 is applied are the following:
_ports/nrf/boards/pca10040/mpconfigboard.mk_:
SOFTDEV_VERSION = 5.0.0

And if you want to use the BLE REPL, _ports/nrf/bluetooth_conf.h_
#define MICROPY_PY_BLE_NUS              (1)

If you want to play with WebBluetooth REPL, i could also recommend you to try out @aykevl's WebBluetooth REPL as well: https://aykevl.nl/apps/nus/

dvsseed, 2018-04-21:
Hi @glennrub,

I really appreciate your kind assistance.

As your step, I modified the files as follow:
boards/s132_5.0.0.ld
drivers/bluetooth/ble_drv.c
boards/pca10040/mpconfigboard.mk
bluetooth_conf.h

then, I did the command $ make BOARD=pca10040 SD=s132 sd...there was no error.
But I tried the https://aykevl.nl/apps/nus/ , and connected the mpus, that it fixed crash when opening NUS Console.

glennrub, 2018-04-21:
Do i understand it correctly that the PR #190 fixes the issue connecting to the NUS console? If so, i will merge the PR into master.

glennrub, 2018-06-14:
PR #197 has now been merged to master, adding SDv6 as default BLE stack. Closing this issue for now. Feel free to re-open it if that PR did not solve the issue.

closed 2018-06-14 by glennrub