tralamazza/micropython #191: Add support for Bluetooth LE stack s132/s140 v6, deprecate s132v2/3/5

glennrub, 2018-04-21:
s132 v6 and s140 v6 are API compatible. This means that support for nrf52832 (s132) and nrf52840 (s140) would be symmetric. 

I proposing to deprecate the support for s132 v2.0.1, v3.0.0 and v5.0.0, and remove all unused BLUETOOTH_SD=130 compile defines. This would make BLE drivers much cleaner, and basically only have 2 variants to switch between compile time, s110 v8 and s132/s140 v6.

One open issue is what to do with feather52 which is bound to s132 v2.0.1. One thing to investigate is if there is an option to do a DFU update to replace the SD.

Other than cleaner code, this will also have a Flash consumption impact as the v6 BLE stack is a bit bigger than v3. 0x26000 (v6) - 0x1F000 (v3) = 28K.

I would really appreciate comments on this.

glennrub, 2018-04-22:
I've started on a branch to make the changes proposed above. I'm still struggling with a FATAL error that occurs during TX of more than one MTU size transaction when using BLE REPL. So for now i will not do a PR. However, here is the working branch if someone wants to take have an early look: https://github.com/glennrub/micropython/tree/s132_s140_v6

glennrub, 2018-05-01:
Work has been pushed as a PR, #197. 

I've tested the updated code on pca10028 (s110 v8), pca10040 (s132 v6)  and pca10056 (s140 v6) using BLE REPL and setting up a second advertiser while in connection. Also, tested s140 in central role doing scanning and connection to a peripheral, also while in connection over BLE REPL. So far it looks good.

However, the current PR breaks feather52 support. I will have to think this a bit through, on how to trigger an upgrade of the feather52 kits to s132v6 over DFU as it might break the on-board bootloader memory setup.

aykevl, 2018-05-01:
I am afraid the bootloader does not support updating a major version of the SD. At least that's what I remember from reading the docs.

Maybe we'll have to continue support s132 v2, unless we can somehow convince Adafruit to update (unlikely I think).

aykevl, 2018-05-01:
Interesting! Yes that would probably be the way to go. It looks like it uses the serial bootloader which probably doesn't rely on the SoftDevice.

aykevl, 2018-05-01:
I don't have a feather52 so I can't test this.

glennrub, 2018-05-01:
I have a feather52 with s132v2.0.1 on it. I will perform some tests on that one. 

Another alternative as i see it, is to treat the feather52 as any other target, requiring external programmer (segger/dap). This way, feather52 is just another board, and from this port point of view, full access to all flash and any the memory layout this port provides.

glennrub, 2018-06-14:
PR #197 merged. Closing issue.

closed 2018-06-14 by glennrub