tralamazza/micropython #194: compilation error for new version

linycm, 2018-04-28:
Hi,

    I encounter many error for the recent version compilation as below. What's the reason?
 GEN build-pca10040/genhdr/qstr.i.last
In file included from ../../py/mphal.h:34:0,
                 from ../../py/mpprint.c:33:
./mphalport.h:31:10: fatal error: nrfx.h: No such file or directory
 #include <nrfx.h>
          ^~~~~~~~
compilation terminate

Rdgs,
Martin

linycm, 2018-04-28:
The environment is: complie gcc-arm-none-eabi-7-2017-q4-major, virtual machine is Vmware workstation 64 bit Ubuntu.

glennrub, 2018-04-28:
Hi @linycm !

Based on the error, not finding nrfx.h, it looks like you have not fetched the nrfx through the git submodule. 
To fetch the submodules you can run the following command inside you local clone of the repository:

`git submodule update --init`

linycm, 2018-04-28:
I had run the command git submodule update --init at the first, but got error. So I run the command git init and then run "git submodule update --init". But don't fetch more submodules. How to do?

glennrub, 2018-04-28:
Can you see the nrfx.h under <clone-root>/lib/nrfx/ ? Or is it missing?

glennrub, 2018-04-28:
one important note on the "git submodule update" is that it has to be done inside the folder you have cloned it to. Else it would not have any effect.

aykevl, 2018-04-28:
> So I run the command git init [...]

Don't do that in an existing repository, although it probably won't do any harm. `git init` is for creating a new repository. From [the docs](https://git-scm.com/docs/git-init#_description):

> Running git init in an existing repository is safe. It will not overwrite things that are already there. The primary reason for rerunning git init is to pick up newly added templates (or to move the repository to another place if --separate-git-dir is given).

linycm, 2018-04-29:
When I download a source.zip file as micropython folder, "git submodule update --init" cannot work. <clone-root>/lib/nrfx/ is empty. When I use git clone... method, "git submodule update --init" works. 

I complie with parameter "FROZEN_MPY_DIR=freeze", the result have error as below:

build-pca10040/frozen_mpy.c:39:26: error: size of 'mp_qstr_const_pool' differ from the size of original declaration [-Werror=lto-type-mismatch]
 extern const qstr_pool_t mp_qstr_const_pool;
                          ^
../../py/qstr.c:98:19: note: 'mp_qstr_const_pool' was previously declared here
 const qstr_pool_t mp_qstr_const_pool = {
                   ^
../../py/qstr.c:113:26: error: size of 'mp_qstr_frozen_const_pool' differ from the size of original declaration [-Werror=lto-type-mismatch]
 extern const qstr_pool_t MICROPY_QSTR_EXTRA_POOL;
                          ^
build-pca10040/frozen_mpy.c:40:19: note: 'mp_qstr_frozen_const_pool' was previously declared here
 const qstr_pool_t mp_qstr_frozen_const_pool = {
                   ^
lto1: all warnings being treated as errors
lto-wrapper: fatal error: /home/martin/Ctools/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc returned 1 exit status
compilation terminated.

aykevl, 2018-04-29:
> When I download a source.zip file as micropython folder, "git submodule update --init" cannot work. 

That's right, when you download the source as a .zip file from GitHub you're only getting the source files, not the git repository. So then all git commands will fail. In most cases, cloning with git is the best option.

> lto-wrapper: fatal error: /home/martin/Ctools/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc returned 1 exit status

It looks like you're using a very new compiler which may give more compile errors. Can you run this command and paste the output?

    arm-none-eabi-gcc --version

And this error:
```
CC modules/uos/microbitfs.c
modules/uos/microbitfs.c: In function 'advance':
modules/uos/microbitfs.c:407:24: error: 'ENOSPC' undeclared (first use in this function); did you mean 'MP_ENOSPC'?
```

It probably shows an error for you as you're using a different toolchain but the fix is as gcc suggests. I'll send a patch soon.

```
modules/uos/microbitfs.c:564:34: error: 'uos_mbfs_mount_obj' defined but not used [-Werror=unused-const-variable=]
STATIC MP_DEFINE_CONST_FUN_OBJ_3(uos_mbfs_mount_obj, uos_mbfs_mount);
```
Interesting. The error is correct but the issue is harmless. Removing this line should silence the error.

aykevl, 2018-04-29:
Oh and so you know: if you add three backticks (```) before and after a code or output block, the code will be much more readable. See [this page](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) for more information.

linycm, 2018-04-29:
Current compiler is 
'''arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]'''.

But I change to arm-none-eabi-gcc 6.3.1 20170620 (release), I got similar error result when compile with '''make FROZEN_MPY_DIR=yourdirectory'''.

aykevl, 2018-04-29:
The link errors haven't been fixed yet.
A temporary workaround is not specifying `FROZEN_MPY_DIR` or removing `-Werror` from the Makefile.
I think it's an issue with how frozen MPY files are generated and should be fixed in MicroPython (not this port).

linycm, 2018-04-30:
@aykevl , @glennrub 
Thank you!
I try lastest git master, it pass the complie when added "'#define MICROPY_HW_HAS_BUILTIN_FLASH (1)"' in mpconfigboard.h.

After I flash firmware inot PCA10040, I put command in Unbutu
'''sudo ampy --port /dev/ttyUSB0 ls''', I cannot got correct AMPY result:
'''
Traceback (most recent call last):
  File "/usr/local/bin/ampy", line 11, in <module>
    sys.exit(cli())
  File "/home/martin/.local/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/martin/.local/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/martin/.local/lib/python3.5/site-packages/click/core.py", line 1066, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/martin/.local/lib/python3.5/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/martin/.local/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/martin/.local/lib/python3.5/site-packages/ampy/cli.py", line 147, in ls
    for f in board_files.ls(directory):
  File "/home/martin/.local/lib/python3.5/site-packages/ampy/files.py", line 103, in ls
    raise ex
  File "/home/martin/.local/lib/python3.5/site-packages/ampy/files.py", line 96, in ls
    out = self._pyboard.exec_(textwrap.dedent(command))
  File "/home/martin/.local/lib/python3.5/site-packages/ampy/pyboard.py", line 265, in exec_
    raise PyboardError('exception', ret, ret_err)
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", in <module>\r\nTypeError: function takes 0 positional arguments but 1 were given\r\n')
'''
   Is it something wrong in my operation for AMPY？

linycm, 2018-04-30:
If I put main.py in freeze folder, compile '''FROZEN_MPY_DIR=freeze''' and removing -Werror from the Makefile, then after I flash firmware, The main.py function can work when ‘’‘import main''' in REPL. But I want main.py function can work when the board reboot. I cannot figure it out.

aykevl, 2018-04-30:
At the moment it appears to be impossible to execute `main.py` as a frozen script. The way I work around it is by using a module with a different name and creating a file in the flash filesystem that imports it:

    >>> f = open('boot.py', 'w')
    >>> f.write('from main import *')
    >>> f.close()

It should be fixed, though...

aykevl, 2018-04-30:
Oh and I see the problem with ampy: the `ls` command doesn't work. Probably because ampy gives it a parameter and this filesystem doesn't support it.
I think other commands will work. I have used it for `put` before.

linycm, 2018-04-30:
@aykevl 
Thank you! The below works. I realize some function when reboot.
>>> f = open('boot.py', 'w')
>>> f.write('from main import *')
>>> f.close()

The command '''put''' file into board works in AMPY.

closed 2018-07-19 by glennrub