tralamazza/micropython #203: LTO issue with upstream

glennrub, 2018-06-06:
In an attempt trying to upmerge with upstream micropython/master the port has become unstable with LTO=1.

Example sequence:
```
>>> from machine import Pin
>>> p = Pin("P17", mode=Pin.OUT)
>>> p.value()
Traceback (most recent call last):
  File "<stdin>", in <module>
AttributeError: 'int' object has no attribute 'value'
>>> p
38398
>>> p = Pin("P17", mode=Pin.OUT)
>>> p
Pin(Pin.cpu.P17, mode=Pin.port=0x0, pin=0x11, )
>>> p.value()
0
>>> 
```
Compiling using LTO=0 and these symptoms seems to disappear.
Needs investigation on what has changed in upstream master that can trigger this issue.

aykevl, 2018-06-07:
That's... strange. It looks a lot like heap corruption, which may of course be triggered by LTO somehow (meaning LTO is not the root cause).

Anyway, if it really is LTO then maybe it's best to just disable it by default. I thought it was a good idea but it may be more trouble than it's worth. And it doesn't even always result in space savings (LTO sometimes increases flash usage due to more inlining across object files).

glennrub, 2018-06-08:
I ran the same merge with LTO=0, then running a quite large python program, and i can still provoke the issue. So i'm not so sure it is related to the LTO anymore. I'm going to start looking at the changes in upstream to see if i can spot something evident.

Renaming the issue title to something more generic.

glennrub, 2018-06-13:
Even if i have not really found the real issue, i believe i have found the patch-set that made it start failing. 
So far i have narrowed it down to be the patches going into upstream/master 10th of May18.

glennrub, 2018-06-18:
After selectively reverting patches, it looks like the issue is due to this commit upstream 7241d90272f4dfe4100723393cc2f348f5d32c0a

Next, would be to figure out why this patch does not work for this port.

glennrub, 2018-06-21:
I think i get a bit closer on this now. The new code which is triggered is a nlr_push, looking at runtime.c function which is called: 
```
void mp_load_method_protected(mp_obj_t obj, qstr attr, mp_obj_t *dest, bool catch_all_exc) {
    nlr_buf_t nlr;
    if (nlr_push(&nlr) == 0) {
        mp_load_method_maybe(obj, attr, dest);
        nlr_pop();
    } else {
        if (!catch_all_exc
            && !mp_obj_is_subclass_fast(MP_OBJ_FROM_PTR(((mp_obj_base_t*)nlr.ret_val)->type),
                MP_OBJ_FROM_PTR(&mp_type_AttributeError))) {
            // Re-raise the exception
            nlr_raise(MP_OBJ_FROM_PTR(nlr.ret_val));
        }
    }
}
```
The old code in repl.c which is currently in master uses mp_load_method_maybe() directly without nlr_push. 

So it might be related to something along the lines of this patch: 5591bd237a69a3a1a6ac03cb1dc9edcde835f708 

One thing i have done is to upgrade my system to arm-gcc 7.2.1.

glennrub, 2018-06-21:
After some debugging and printf's it looks like nlr_push always returns 1 in this function. Hence, it will jump to the else clause. The inner "if" will not run as catch_all_exc is set to true in all cases in the repl.c (mp_repl_autocomplete). In practice it looks like no code is actually executed in this function.

glennrub, 2018-07-04:
Solved.

closed 2018-07-04 by glennrub